Dharakverma/vehicle_control_system #3: Team 23 Review: Hazard Analysis

LemonU-mcmaster, 2022-10-23:
Honestly the document is very well-made and I'm having a hard time trying to spot issues, but anyways:
1. SR1 and SR3 do kind of overlap. Specifically, SR1 sounds like a superset of SR3.
2. It would be ideal to give a symbolic parameter for the _"safe operational temperature"_ specified in SR2

smiths, 2022-10-23:
@LemonU-mcmaster, please create a separate issue for each of your issues.  :smile:

LemonU-mcmaster, 2022-10-23:
> @LemonU-mcmaster, please create a separate issue for each of your issues. 😄

Thanks for the notice. I've divided them into two issues.

Dharakverma, 2022-12-09:
There is no SR3.

abhishekmagdum, 2022-12-09:
SR3 does not exist. I assume you mean the two separate SR2's? 

We've separated them since they affect different subsystems. One can brown out the accumulator, while the other affects the internal temperature, which is a much bigger issue (safety).

closed 2022-12-09 by abhishekmagdum