Dharakverma/vehicle_control_system #6: Team31 Review: Introduction + Security and Usability NFR Tests Comments

chrisfandrade16, 2022-11-04:
Regarding the introduction/summary:
- I found it a bit confusing to understand what the project is from a high-level perspective after reading the introductory sections, perhaps the summary paragraph could've been slightly longer
- I am aware that I can read other documentation, which I did and it provided more clarity, but I think the summary should at least provide me with a basic understanding of what's going on so I can understand the tests more easily without flipping through other documents

Regarding security tests:

- The paragraph in this section mentions that no security tests will be neccessary as the hardware is out of the team's hands, but I think there are security issues that should be addressed that do not only pertain to the GitHub source code privatization which was mentioned
- For example, is it possible the source code can be extracted or modified before/during/after implementation on the hardware, causing dangers? Could the control system logic cause issues or failures to the hardware (instead of vice versa), which could then propagate in the system and cause more errors? Is there anything the Formula drivers should know beforehand regarding their safety and security? How can their security be verified?

Regarding usability tests:

- I noticed there were few usability requirements, so this may have been the reason, but I believe some kind of tests with different types of hardware, at least one, would be necessary to fully validate the product
- Although the hardware is not the part you are designing, the control system will ultimately be used on hardware, and so it should probably be tested on different kinds to verify the control system is usable and safe despite variations on the hardware

smiths, 2022-11-05:
@chrisfandrade16, for the next time you do this, please create one issue per "issue".  It is easier to discuss and close issues when they are self-contained. :smile:

chrisfandrade16, 2022-11-05:
> @chrisfandrade16, for the next time you do this, please create one issue per "issue".  It is easier to discuss and close issues when they are self-contained. :smile:

@smiths Oh ok thank you for the reminder, will do. Got carried away on this one!

closed 2023-04-07 by abhishekmagdum